pokusew/nfc-pcsc #133: Error when writing data. Receives an Invalid data length error.

TommyLiang1, 2022-07-20:
Hi @pokusew.

I am testing my NFC card reader through the command ```npm run example```. I am running into an issue when writing to a MiFare Classic 1K card. I was following issue #25 as I received a "Write operation failed: Status code: 0x6300" error at first.

After passing in the 3rd parameter for the reader.write() function so that it looks like this ```await reader.write(4, data, 16);```, I receive an "Invalid data length" error.

This is the if statement that returns that error
[image]
I added a console.log line to print the values of data.length, blockSize and data.length%blockSize.
It returns 4, 16, 4 respectively.
Since the data.length is 4 and blockSize is 16, the if statement conditions will be met and will give an "Invalid data length" error as I stated.

Why is this the case? Why is data.length returning 4(I'm assuming that it should return multiples of 16 in order to bypass that if statement)?

I tried commenting out the if statement and the last line in the image replaced the error.
[image]
I know that this is wrong as the very next if condition won't run as data.length is not greater than blockSize which is why I made this issue.

**Environment Information:**
OS: Ubuntu 20.04
Node Version: 16.16.0
NPM Version: 8.11.0
Card Reader: ACR122U

TommyLiang1, 2022-07-20:
Hi @pokusew,
Thank you for your quick response.

After reading your comment, I traced back to where the data variable is initialized.

It is through this line ```const data = Buffer.allocUnsafe(4).fill(0);``` where the data variable is initialized before the reader.write() call. I changed the 4 to 16 to ...Buffer.allocUnsafe(16)... and it works.

Is this an appropriate fix?

pokusew, 2022-07-20:
@TommyLiang1, Yes, it is (I assume you are talking about [line 75 in read-write.js example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/read-write.js#L75)).

BTW, you can look directly at the [mifare-classic.js example](https://github.com/pokusew/nfc-pcsc/blob/master/examples/mifare-classic.js#L152) where it is already set up correctly for MIFARE Classic (you can use `npm run example-mifare-classic` to run it).

Hope it helps. 🙂 

---

PS Don't forget to star ⭐️my library, if you find it useful. 😃Thanks.