pokusew/nfc-pcsc #138: was compiled against a different Node.js version using NODE_MODULE_VERSION 111. This version of Node.js requires NODE_MODULE_VERSION 110.

odysahe, 2023-01-13:
Hello, sorry I'm still a newbie, I created a nuxtjs and electron desktop application project for attendance list.
My nfc reader uses `ACR122`.
After i installed [nfc-pcsc](https://github.com/pokusew/nfc-pcsc) i got the following error:

```
A JavaScript error occurred in the main process 
Uncaught Exception:
Error: The module '/absdesk/node_modules/@pokusew/pcsclite/build/Release/pcsclite.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 111. This version of Node.js requires
NODE_MODULE_VERSION 110. Please try re-compiling or re-installing
```
I have changed node version, after browsing to [node.js release](https://nodejs.org/en/download/releases/) website i am not getting `NODE_MODULE_VERSION 110`.

[image]

I've also tried the solution from [pokusew/node-pcsclite](https://github.com/pokusew/node-pcsclite/issues/38) issues but it still doesn't work.
can someone help me?

**node.js v19.4.0**
**nfc-pcsc : ^0.8.1**
**nuxt : ^2.15.8**
**electron : ^22.0.0**
**electron-builder : ^23.6.0**
**electron-devtools-installer : ^3.2.0**

davidgs, 2023-04-05:
Solution: 

`yarn add -D @electron/rebuild`

Then add the following to your package.json: 

```
"rebuild": "electron-rebuild --parallel --types prod,dev,optional --module-dir release/app",
```

You can then run `yarn run rebuild` and it will rebuild with the right node version.

mattlebel, 2023-05-31:
I was running into this error:

```
App threw an error during load
Error: The module '/Users/USER_VAR/electron-quick-start/node_modules/@pokusew/pcsclite/build/Release/pcsclite.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 108. This version of Node.js requires
NODE_MODULE_VERSION 116. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:2:1822)
    at Module._extensions..node (node:internal/modules/cjs/loader:1354:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:2:1822)
    at Module.load (node:internal/modules/cjs/loader:1124:32)
    at Module._load (node:internal/modules/cjs/loader:965:12)
    at f._load (node:electron/js2c/asar_bundle:2:13330)
    at Module.require (node:internal/modules/cjs/loader:1148:19)
    at require (node:internal/modules/cjs/helpers:110:18)
    at bindings (/Users/USER_VAR/electron-quick-start/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/Users/USER_VAR/electron-quick-start/node_modules/@pokusew/pcsclite/lib/pcsclite.js:4:37)
[60615:0531/124805.752891:ERROR:CONSOLE(1)] "Uncaught (in promise) TypeError: Failed to fetch", source: devtools://devtools/bundled/panels/elements/elements.js (1)

```

I solved it by:

1. Updating my init scripts in `package.json` to:

```
  "scripts": {
    "start": "electron .",
    "ebuild": "electron-rebuild",
    "build-workspace": "npm install && npm run ebuild"
  },
```

2. Installing `electron/rebuild` via

`npm install --save-dev @electron/rebuild`

3. Running `npm run build-workspace`

alessandroraffa, 2024-03-07:
This is my issue:

```log
Error: The module '/projectRoot/node_modules/@pokusew/pcsclite/build/Release/pcsclite.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 115. This version of Node.js requires
NODE_MODULE_VERSION 121. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (node:electron/js2c/node_init:2:2214)
    at Module._extensions..node (node:internal/modules/cjs/loader:1343:18)
    at Object.func [as .node] (node:electron/js2c/node_init:2:2214)
    at Module.load (node:internal/modules/cjs/loader:1098:32)
    at Module._load (node:internal/modules/cjs/loader:945:12)
    at c._load (node:electron/js2c/node_init:2:13672)
    at Module.require (node:internal/modules/cjs/loader:1122:19)
    at require (node:internal/modules/helpers:130:18)
    at bindings (/projectRoot/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/projectRoot/node_modules/@pokusew/pcsclite/lib/pcsclite.js:4:37)
```

I've added and run the rebuild script (also cleaning the project):

```json
"rebuild": "electron-rebuild --parallel --types prod,dev,optional --module-dir release/app",
```

But the issue persists.

Below my environment:

macOS: Ventura 13.3.1
nvm: 0.38.0
node: v20.11.0
npm: 10.2.4
yarn: 4.1.0

electron: 29.1.0
nfc-pcsc: 0.8.1
@electron/rebuild: 3.6.0
electron-builder: 24.13.3
electron-devtools-installer: 3.2.0
electron-icon-maker: 0.0.5
electron-reload: 2.0.0-alpha.1