pokusew/nfc-pcsc #139: UnhandledPromiseRejectionWarning: WriteError: Write operation failed: Status code: 0x6300

riegeler, 2023-03-22:
Hello,

I'm using nfc-pcsc library with an ACR122U connected to a Systec CTR-500 PLC running Node-RED (=node.js).
My NFC card is integrated into the Nordic µC nRF52840 and it seems to be the same as NXP NTAG213/215/216 which is **not** a Mifare.
It's formatted to NFC Type 2.
Reading and writing in principle works. The node-red-contrib-nfc-pcsc includes your nfc-pscs:
`const { NFC } = require("nfc-pcsc");`

I don't know what happened but since a few days after calling
`await reader.write(4,outData);`
I get the error:

```
(node:268) UnhandledPromiseRejectionWarning: WriteError: Write operation failed: Status code: 0x6300
	at ACR122Reader.write (/root/.node-red/node_modules/nfc-pcsc/dist/Reader.js:493:13)
(node:268) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 5)
```
Reader.js:493:13 is only printing the error:
```
if (statusCode !== 0x9000) {
      throw new _errors.WriteError(_errors.OPERATION_FAILED, `Write operation failed: Status code: 0x${statusCode.toString(16)}`);
}
```

However the data is written correctly, I can't recognize any error. But I'd like to understand the error message. What could lead to error 0x6300?

The capability container CC3 of the Type 2 header is set to 0x00 which means read and write access granted without any security.

I'd be happy if some could give me a hint for this error.
If any information is missing don't hesitate to ask me.

Thanks a lot!

davidgs, 2023-04-05:
I'm now stuck with this `Write operation failed: Status code: 0x6300` I'm trying to write an NDEF message to a mifare classic 1K card. I can authenticate to each sector and read just fine. I then re-authenticate to each sector to write to it, but ... fail. I'm stumped.

riegeler, 2023-04-20:
I implemented a try-catch around 
`await reader.write(4,outData);`
and indeed get this error 0x6300 in catch section.

But happens all the time regardless if I write successfully or if I abort write process.
How can I catch an aborted write process?

riegeler, 2023-05-04:
Furthermore I get this error 0x6300 also during read:

```
May  4 13:49:55 ctr-700-456234 node-red[8581]: (node:8581) UnhandledPromiseRejectionWarning: ReadError: Read operation failed: Status code: 0x6300
May  4 13:49:55 ctr-700-456234 node-red[8581]:     at ACR122Reader.read (/root/.node-red/node_modules/nfc-pcsc/dist/Reader.js:432:13)
May  4 13:49:55 ctr-700-456234 node-red[8581]: (node:8581) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
```

In Read.js at line 432 there is only the error printing:
```
419    try {
420      response = await this.transmit(packet, length + 2);
421    } catch (err) {
422      throw new _errors.ReadError(null, null, err);
423    }
424
425    if (response.length < 2) {
426      throw new _errors.ReadError(_errors.OPERATION_FAILED, `Read operation failed: Invalid response length ${response.length}. Expected minimal length is 2 bytes.`);
427    }
428
429    const statusCode = response.slice(-2).readUInt16BE(0);
430
431    if (statusCode !== 0x9000) {
432      throw new _errors.ReadError(_errors.OPERATION_FAILED, `Read operation failed: Status code: 0x${statusCode.toString(16)}`);
433    }
```

Is 0x6300 only a general error of ACR122U independend of reading or writing? Is it possible to get some more details about it?