pokusew/nfc-pcsc #143: Module not found: Error: Can't resolve 'fs'

sunnyso246, 2023-06-22:
Hi,

 I add the library in my react project and always get this error. Please advise how to solve it. 

ERROR in ./node_modules/bindings/bindings.js and Module not found: Error: Can't resolve 'fs' in node-modules.

Use react script 5.0.1 and webpack 5.

Thanks
Sunny.

danielduan, 2023-09-20:
I'm seeing this problem as well when building for Electron. Is there a solution for this?

alexgoto, 2024-02-07:
> Hi,
> 
> I add the library in my react project and always get this error. Please advise how to solve it.
> 
> ERROR in ./node_modules/bindings/bindings.js and Module not found: Error: Can't resolve 'fs' in node-modules.
> 
> Use react script 5.0.1 and webpack 5.
> 
> Thanks Sunny.

`fs` is is Node.js provided context. React works inside browser, where Node.js context is unavailable.

This module works only in Node.js.

If we talk about the whole task - browser can not provide access to Smart Cards from JavaScript (React in your situation) directly. Is is possible only if host machine of user of your JavaScript provide some of interface for this. For example this interface is browser [extension](https://github.com/cardid/webcard)