pokusew/nfc-pcsc #144: Unable to run the basic example

anotherrohit, 2023-06-22:
I found this package and attempted to install it. I immediately was hit by an error as below:
C:\Users\nfc-pcsc-master\nfc-pcsc-master>npm run example

> nfc-pcsc@0.8.1 example
> node -r @babel/register examples/read-write.js

node:internal/modules/cjs/loader:1078
  throw err;
  ^

Error: Cannot find module '@babel/register'
Require stack:
- internal/preload
    at Module._resolveFilename (node:internal/modules/cjs/loader:1075:15)
    at Module._load (node:internal/modules/cjs/loader:920:27)
    at internalRequire (node:internal/modules/cjs/loader:174:19)
    at Module._preloadModules (node:internal/modules/cjs/loader:1431:5)
    at loadPreloadModules (node:internal/process/pre_execution:606:5)
    at setupUserModules (node:internal/process/pre_execution:118:3)
    at prepareExecution (node:internal/process/pre_execution:109:5)
    at prepareMainThreadExecution (node:internal/process/pre_execution:37:3)
    at node:internal/main/run_main_module:10:1 {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ 'internal/preload' ]
}

Node.js v18.15.0

Any luck with this issue?

newo27, 2023-08-08:
It looks like you just have to install babel

hope this helps!

newo27, 2023-08-08:
you can do this by using npm
npm i @babel/node